adafruit/Adafruit_CircuitPython_RFM9x #9: rfm9x_simpletest.py fails under Blinka

jerryneedell, 2018-08-13:
I tried running try rfm9x_simpletest.py on a RaspberryPi 3B+ with the latest Blinka (0.2) - the SPI stuff seems fine, but I get the following error
```
pi@gjnpi3p-1:~/projects/rfm9x/orig $ python3.6 rfm9x_simpletest.py
Traceback (most recent call last):
  File "rfm9x_simpletest.py", line 39, in <module>
    rfm9x.send('Hello world!\r\n')
  File "/usr/local/lib/python3.6/site-packages/adafruit_rfm9x.py", line 561, in send
    self._write_from(_RH_RF95_REG_00_FIFO, data)
  File "/usr/local/lib/python3.6/site-packages/adafruit_rfm9x.py", line 414, in _write_from
    device.write(buf, end=length)
  File "/usr/local/lib/python3.6/site-packages/busio.py", line 112, in write
    return self._spi.write(buf, start, end)
  File "/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/raspi_23/spi.py", line 47, in write
    self._spi.writebytes([x for x in buf[start:end]])
TypeError: Non-Int/Long value in arguments: 76a38380.
```

By making this change - the error goes away

```
39c39,40
< rfm9x.send('Hello world!\r\n')
---
> msg = str.encode('Hello worlt!\n\r')
> rfm9x.send(msg)
```

I have not fully tested that this is transmitting and receiving. I am testing it remotely and only have one rfm9x connected.  I'll do more testing next weekend.  The original flle runs OK und CircuitPython -- or it did last time I tried it.

jerryneedell, 2018-08-28:
Yes-resolved by #10.

closed 2018-08-28 by jerryneedell

ladyada, 2018-08-28:
awesome