adafruit/Adafruit_CircuitPython_RFM9x #27: RFM9X RESET not working with Raspberry Pi 4

jerryneedell, 2019-07-02:
I have been tryinog to  use an RFM9x on a Raspberry Pi 4 with Raspian "Buster"
At first I kept getting the message that it could not verify the device ID but I checked the wiring and it is OK.  ( I use the same wiring with an RFM69 and it works)

The issue appears to be related to the toggling of the RESET which is different for the RFM9x vs RFM69

If I disconnect the RESET line then it can deter the chip ID and it receives packets, to appear to miss many packets as well.

I also tried it with the TinyLoRa library which is xmit only. It works OK as long as the RESET line is not connected (RESET is not used for TinyLoRa)

I have a lot more digging to do , but just wanted to post this as an alert and see if anyone else can reproduce the issue.

I'll update when I can do more testing.

Note: The rfm9x library and RESET work on a Raspberry Pi Zero-W with Raspian Stretch

ladyada, 2019-07-02:
https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/master/adafruit_rfm9x.py#L361

pullups aren't working yet on pi 4 so you won't be able to get it working till its fixed downstream :)

https://sourceforge.net/p/raspberry-gpio-python/tickets/168/

jerryneedell, 2019-07-02:
Whew! Thank you! You’ve saved my remaining hair!

jerryneedell, 2019-07-21:
This has been fixed by a recent update to RPI.GPIO
https://sourceforge.net/p/raspberry-gpio-python/tickets/168/
tested with rfm9x and tinylora libs with RESET connected on RPi 4.

closed 2019-07-21 by jerryneedell

ladyada, 2019-07-21:
yayyyy