adafruit/Adafruit_CircuitPython_RFM9x #31: I believe the RSSI function is returning the wrong number.

nlbutts, 2019-11-26:
On line 553 the RSSI register (for Lora) is read and 137 is subtracted from the value. During testing I seem to lose the LORA signal at -101 dBm. Looking at the Semtech data sheet (section 5.5.5) the formula for RSSI is:
` RSSI (dBm) = -157 + Rssi, (when using the High Frequency (HF) port)
or
RSSI (dBm) = -164 + Rssi, (when using the Low Frequency (LF) port)
` 

Working the numbers back when this library reads -101, the register would have contained 36, running the correct formula gives me an RSSI of -121 dBm. I'm running the Lora radio at 125 KHz bandwidth and a SF of 7, which should give me a theoretical -125 dBm receive sensitivity. This seems in the realm of possibility. 

Am I missing something or is the number incorrect in the code?

brentru, 2019-11-26:
@nlbutts
The library code uses RFM9x Datasheet, which seems like it doesn't take HF/LF into account like the SX1276 formula (https://www.digikey.com/en/datasheets/rf-solutions/rf-solutions-rfm95_96_97_98w):

[image]

`        return self._read_u8(_RH_RF95_REG_1A_PKT_RSSI_VALUE) - 137`

I'd be open to implementing the RSSI formulas from the SX datasheet in this library, unless you'd want to take this on and open a PR?

nlbutts, 2019-11-27:
It appears Hoperf fixed the datasheet. I just downloaded Version 2.0 of the datasheet from their web-site and it shows the same formula as Semtech's datasheet. 

Excuse my ignorance, but what is a PR?

brentru, 2019-11-27:
@nlbutts Cool - confirmed the latest hoperf DS reflects the semtech formulas. (https://www.hoperf.com/data/upload/portal/20190801/RFM95W-V2.0.pdf).

PR as in a pull request to this repository. Alternatively I can update the code and library to reflect the issue

kattni, 2020-05-04:
@brentru Is this something you're still working on?

nlbutts, 2020-05-08:
I wanted to use the SX127x parts in both LoRa and FSK mode, so I switched to the UPM code base.

brentru, 2020-12-21:
@jerryneedell  Was this ever implemented in any of the work you've done for RFM9x?

jerryneedell, 2020-12-21:
@brentru I have not but it look like a good  thing to do -- there are also some proposed changes in #51 that I have been meaning to test and implement. Unfortunately, I have not had much time available recently. That should be improving this week.
If you want this done immediately, then you are probably better off doing it yourself. I hope to be able to get to this later this week or next week.

jerryneedell, 2020-12-22:
looks like this will work...unless I am missing somethng.
```
    @property
    def rssi(self):
        """The received strength indicator (in dBm) of the last received message."""
        # Read RSSI register and convert to value using formula in datasheet.
        # Remember in LoRa mode the payload register changes function to RSSI!
        rssi = self._read_u8(_RH_RF95_REG_1A_PKT_RSSI_VALUE)
        if self.low_frequency_mode:
            rssi -= 157
        else:
            rssi -= 164
        return rssi
```

brentru, 2020-12-22:
@jerryneedell  Yeah, checked the v2 of the hoperf datasheet and the code in this library looks correct to me.

jerryneedell, 2020-12-22:
It has not been updated yet

jerryneedell, 2020-12-22:
@brentru I'm confused. Are you saying it does not need to be changed?

brentru, 2020-12-22:
@jerryneedell  Sorry - I misread your post as a code snippet from `master`! The code you posted looks good & matches the Semtech and hoperf datasheets.

jerryneedell, 2020-12-22:
@brentru I was planning to combine this with #51. Is that OK or would you prefer separate PR's?

brentru, 2020-12-22:
@jerryneedell  That's OK - fold it in!

jerryneedell, 2020-12-24:
fixed by #57

closed 2020-12-28 by brentru