adafruit/Adafruit_CircuitPython_RFM9x #32: Getting Runtime Error

darkpanda08, 2019-12-10:
Hello,
I was trying out the library using the tutorial given on Adafruit Website with Adafruit RFM95W LoRa Radio Transceiver Breakout - 433MHz  (https://learn.adafruit.com/lora-and-lorawan-radio-for-raspberry-pi) with Raspberry Pi 3B+ running Raspbian Buster.

Upon running just the following code:

```
import digitalio
import board
import busio
import adafruit_rfm9x

CS = digitalio.DigitalInOut(board.CE1)
RESET = digitalio.DigitalInOut(board.D25)
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, 433.0)
```
I got the following error:
```
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, 433.0)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_rfm9x.py", line 367, in __init__
    raise RuntimeError('Failed to find rfm9x with expected version -- check wiring')
RuntimeError: Failed to find rfm9x with expected version -- check wiring
```

I checked my wiring but still the same error was coming.
Please help me with this.

brentru, 2019-12-10:
Hi darkpanda08, 

Could you please post a photo clearly showing your wiring?

darkpanda08, 2019-12-10:
Thanks for your reply,
I have connected like this:
VIN     --- 3.3V
GND   --- GND
G0      --- GPIO 5
RST    --- GPIO 25
SCK    --- SCK
MISO --- MISO
MOSI --- MOSI
CS     --- CE1 

My wires are not arranged properly, hence I didn't post the pic, If required I will try to post that also.

ladyada, 2019-12-10:
post a picture

darkpanda08, 2019-12-10:
[image]

ladyada, 2019-12-10:
OK for tech support please post in the adafruit forums - its probably a soldering issue :)

darkpanda08, 2019-12-10:
You mean soldering of headers with breakout board ?

ladyada, 2019-12-10:
yes

darkpanda08, 2019-12-10:
Okay, I will check that.

I'm unable to post on Adafruit forum, it says only available for Adafruit customers. I purchased this breakout board from Arrow Electronics.

ladyada, 2019-12-10:
you need to make an account - the board is made by us

darkpanda08, 2019-12-10:
Okay, Thank You

closed 2019-12-10 by darkpanda08