adafruit/Adafruit_CircuitPython_RFM9x #34: Implement "reliable datagram"

jerryneedell, 2019-12-16:
The Radiohead library has a "Reliable Datagram" mode taht uses acknowledgement and retries to insure packet delivery.

Is there interest in implementing that here. I'll be happy to take it on, but wanted to see if there was any concern or interest.

jerryneedell, 2020-01-27:
This is just to update the status of "reliable datagram".
The good news is that the basic function is working but the bad news is that the timing performance with Circuitpython and especially on the Raspberry Pi  is making it very difficult to get ti to work reliably with boards using the Arduino/Radiohead library. I am having a lot of trouble getting the Circuitpython library to be able to respond fast enough to receive the "ack" packets.
I added a workaround to the Circuitpython side to allow a delay to be specified, but this does not help when talking to Arduino/Radiohead. 

I'm still working on this -- trying to find ways to make it respond better.

jerryneedell, 2020-02-06:
Sorry for the delay. It will likely be another 2 weeks before I can get back to working on this.

lgnashold, 2020-02-19:
I think I would be interested in using and/or helping implementing this.

marc-gist, 2020-02-21:
I would love this... i can get the RadioHead/Arduino to talk to the Pi, but sending data from the Pi to RadioHead never works :(

Metallicats44, 2020-02-21:
I was having the same issue.. Try including the following on the Pi code after instantiating rfm9x.. it worked for me:
rfm9x.enable_crc = True

jerryneedell, 2020-05-01:
implemented with #42

closed 2020-05-01 by jerryneedell