adafruit/Adafruit_CircuitPython_RFM9x #38: Clarify the 4 bytes added to the beginning of each buffer

mcsarge, 2020-02-21:
In working with the library and using it to send to an ESP based LoRa device that does not use this library, I found that this library adds 4 bytes to the start of every buffer for compatibility with RadioHead. I found this extra 4 bytes were being added by examining the code. 
Solutions:
1. Add a setting that turns this compatibility off and on (preferred)
2. Clarify in the examples (and in the header of the adafruit_rfm9x.py file that when sending to or receiving from devices that are not RadioHead compatible, they will need to account for it.

jerryneedell, 2020-02-21:
@mcsarge  Good suggestions -- This library was intended to be Radiohead compliant, but clarification is always a good idea. There is a parameter that allows the user to select if the header is removed from the packet before it is passed back to the user code on receive but that won't help with the transmitted packet. Adding an option to remove the header will require a few other changes so the receiver will accept packets less than 5 bytes. Is there a minimum packet size in general? Not a big deal, but something to take into account. I can't think of a "downside' to adding the option for go 'leaderless' but would like to hear more thoughts on it.

mcsarge, 2020-02-21:
Probably only small packets will be used when doing tests, like I did, On receive, if you configure no header, why would it care what the buffer size is and on transmit, if header is off, still do not care. But if header is on for Tx, it should require >4 bytes and on Rx, I guess drop any buffer <=4.

jerryneedell, 2020-02-21:
> Probably only small packets will be used when doing tests, like I did, On receive, if you configure no header, why would it care what the buffer size is and on transmit, if header is off, still do [...]

No reason to care if no header, but I was just pointing out that it requires additional changes to the code. Not a problem.

mcsarge, 2020-02-21:
I can provide an example using the CubeCell board that accounts for the extra 4 bytes or if you change it an example both ways....

jerryneedell, 2025-02-17:
The ability to remove the RadioHead Header has been implemented in the Adafruit_CircuitPython_RFM library
https://github.com/adafruit/Adafruit_CircuitPython_RFM.git
Closing since this will not be implemented here  to avoid adding code to the already constrained build for the feather_m0_rfm9x board.