adafruit/Adafruit_CircuitPython_RFM9x #43: Receiving all zeroes eventually

tahnok, 2020-04-27:
Hello!

I am running into an issue where after successfully receiving for a while, eventually I receive all 0s, on any call to `receive`

I tried calling the `reset` method when this occurs, but it persists anyway. The only solution I've found is to restart the little script I have (which indicts it's not a transmit problem)

Program here: https://gist.github.com/tahnok/dad135976baead3b38fb074ff500dfa6

I can try and debug, but I'm not really sure how to get more useful information...

jerryneedell, 2020-04-27:
How long is “a while”?
Is this on a Raspberry Pi?
What rfm9x board are you using?
How large is the packet of all zeros?

tahnok, 2020-04-27:
> How long is “a while”?
unclear, at least 15 minutes?
> Is this on a Raspberry Pi?
Yep, rpi 2
> What rfm9x board are you using?
rfm95w, from aliexpress
> How large is the packet of all zeros?
Not sure, I'll grab that next time it goes 💥

tahnok, 2020-04-27:
It's returning `bytearray(b'\x00\x00\x00\x00\x00\x00\x00')` (8 bytes)

jerryneedell, 2020-04-27:
Shouldn’t there be some indentation here?
https://gist.github.com/tahnok/dad135976baead3b38fb074ff500dfa6#file-bastion-py-L6
It is not clear how you are handling a None response.

tahnok, 2020-04-27:
Must have messed up creating the gist, sorry about that

https://gist.github.com/tahnok/4b5c88106a61da452ae948bf6dd82a68

Interestingly, it has been running all night with the transmitter off and this issue hasn't cropped up

jerryneedell, 2020-04-27:
What is the Transmitter?

tahnok, 2020-04-28:
A SAMD21 board I made, and my own hand rolled driver for the rfm9x, so it's possible I'm transmitting some garbage accidentally?

https://github.com/tahnok/hummingbird_atmel_studio/blob/master/Hummingbird/rfm9x.c

jerryneedell, 2020-04-28:
i get a "404" error on that link.
I have not been able to reproduce this issue.  I'll run a "listener" on my Raspberry Pi for a few days and see if it does anything similar.
i do see many random packets of "stuff" but so far, not repeated sets of 0x00 s

tahnok, 2020-04-28:
Whoops, I didn't realize that was a private repo, it's public now

jerryneedell, 2020-04-28:
I just ran my Raspberry Pi overnight just listening and did not see any packets of 0x00's.
I'm not sure how to help track this down without more information to go on.

tahnok, 2020-05-04:
Ok, thanks for investigating! I'll see if I can reproduce on different hardware or with a different cable steup

closed 2020-05-04 by tahnok