adafruit/Adafruit_CircuitPython_RFM9x #52: No device detected on SPI0 with CE0

brainelectronics, 2020-11-03:
Hi together,
I'm using a RFM95 module directly connected via jumper cables to a Raspberry Pi 3B+ (Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux) running Raspbian GNU/Linux 10 (buster)

I've followed the steps of installation in this readme and the [Adafruit LoRa learning page](https://learn.adafruit.com/lora-and-lorawan-radio-for-raspberry-pi/)

Everything is working nice and smooth with CE1/D7 or e.g. D25 but CE0/D8 fails. May I missed to enable something in the `/boot/config.txt`?

Logs and used code below

`pip freeze`
```bash
Adafruit-Blinka==5.6.0
adafruit-circuitpython-busdevice==5.0.1
adafruit-circuitpython-rfm69==2.1.0
adafruit-circuitpython-rfm9x==2.0.4
Adafruit-PlatformDetect==2.20.0
Adafruit-PureIO==1.1.7
pkg-resources==0.0.0
pyftdi==0.52.0
pyserial==3.4
pyusb==1.1.0
rpi-ws281x==4.2.4
RPi.GPIO==0.7.0
spidev==3.5
sysv-ipc==1.0.1
```

`cat /boot/config.txt`
```bash
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
```

`cat rfm9x_check.py`
```python
import time
import busio
from digitalio import DigitalInOut, Direction, Pull
import board
import adafruit_rfm9x

# Configure RFM9x LoRa Radio
# CS = DigitalInOut(board.CE1) # works
# CS = DigitalInOut(board.D7) # works

CS = DigitalInOut(board.D25) # works

# CS = DigitalInOut(board.CE0) # no device detected
# CS = DigitalInOut(board.D8) # no device detected

RESET = DigitalInOut(board.D24)
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)

# Attempt to set up the RFM9x Module
try:
    rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, 868.0)
    print('RFM9x: Detected')
    exit()
except RuntimeError as error:
    # Thrown on version mismatch
    print('RFM9x Error: ', error)
    exit()
```
`RFM9x Error:  Failed to find rfm9x with expected version -- check wiring`

jerryneedell, 2020-11-03:
Take a look at the warning about using CE0 here:

https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices

brainelectronics, 2020-11-06:
Thanks for that link! Should have read the docs for CircuitPython as well...

closed 2020-11-06 by brainelectronics