adafruit/Adafruit_CircuitPython_RFM9x #53: Interoperability between SX1262 and SX1272

TinManAkshay, 2020-11-24:
Hi,

I have been using these libraries for so long now on dragino lora hat on Pi. Earlier we used to have dual communication between Adafruit atmega 32u4 and dragino lora hat.

Now we have our custom board which has new Semtech SX1262 on it and it can successfully transmit data packets to SX1272 (using these libraries). But when I try to send data back to SX1262, it doesnt receive it.

Config on both sides -> 923.3mhz, sf=7, cr=4/5, power=22, BW=500khz, CRC is on. 

I am wondering what could be the reason of SX1262 not receiving the data packet from SX1272. Does 4 byte header compatible with RadioHead's implementation make a difference like I am trying to extract the 4-byte header like this way:

[image]

jerryneedell, 2020-11-24:
I'm sorry, but I don't understand the "big picture" of your configuration. It is not clear to me where you are using the CircuitPython code. I assume it is on the Raspberry Pi? Is the 32U4 using the Radiohead Library? and is configured for "reliable datgram" or simple "datagram"?
 Since the example you have is using "with_ack=True" I assume you are using "reliable datagram" so every packet transmitted must be "acked". The Circupython library cannot always Ack a packet from Radiohead reliably, especially on a Raspberry Pi just due to the timing constraints.  

Since you've been using this "a long time" and the "reliable datagram" support on CircuitPython is relatively new, I wonder if you have inadvertently added complication you don't want.

I'm trying to understand what has changed for you. If you can clarify that, I can try to help.

TinManAkshay, 2020-11-24:
Hi @jerryneedell ,

I meant to say that we use Radiohead library on Pi and LoRa.h on Atmega 32u4 module. As you know they both have SX1272 chips and they both are communicating with each other without any doubts. 

So here is the situation arises we have our own board which has SX1262 chip on it and we still use Radiohead library on Pi. I want them to communicate with each other without any flaws. So far SX1262 can transmit data to sx1272 and sx1272 receives the same data, but other way around doesn't work like when I try to send the data back to SX1262, Sx1262 doesn't receive it. 

I wonder why sx1262 is not receiving the data. Are you familiar with SX1262 working environment?

Thanks,
Akshay

jerryneedell, 2020-11-24:
Hello Akshay,
I'm sorry if I confused things. I have had no experience with the SX1262 chips so I don't think I can be of much help to you.  Have you verified that your code for the SX1262 board works for both transit and receive? That is, you said you can transmit from the SX1262 to the SX1272 but not receive.  Have you been able to try receiving data from another SX1262 board?  

The example you posted above is from the CircuitPython examples but you said you are using the RadioHead library on the Raspberry Pi, so I was confused. The CircuitPython code was written to be compatible with the RadioHead Library so a board using the RadioHead Library with Arduino can communicate with a board using CircuitPython but both are for the SX127x chips. In order to be compatible, the CircuitPython library uses the same 4 byte header as the RadioHead Library.  See - https://www.airspayce.com/mikem/arduino/RadioHead/ 

Good luck!
Jerry

TinManAkshay, 2020-11-24:
Hi,

Yes I have two SX1262 board communicating with each other perfectly fine. Also, I got SX1272 and SX1262 working with each other as well. Both of them has arduino based libraries.
Now I want to get this library working with arduino based library for SX1262. I saw when SX1262 transmits the packet and this SX1272 library receives the packet. But when I try to work other way around, SX1262 doesnt receive it. So I am wondering how I can make this library working with SX1262 like dual communication.
Just out of curiosity, I noticed in these libraries that it is mentioned that there is no sync word set for LoRa module. Is there anyway we can set 0x34 (Sync word for public)?

closed 2020-12-07 by TinManAkshay