adafruit/Adafruit_CircuitPython_RFM9x #55: Add FSK/OOK compatibility to Driver

lgnashold, 2020-12-14:
I am working on using the RFM98PW on a satellite, and as part of that, I will need to leverage the OOK capabilities of the radio module to transmit an easily-understood beacon. To do this, I have been working on adding FSK/OOK capabilities to the RFM9x driver. 

I have been working on a barebones implementation [on a fork](https://github.com/lgnashold/Adafruit_CircuitPython_RFM9x), although it needs more testing and probably more configuration. I have been mostly targeting OOK, although it shares a lot of overlap with FSK. 

I have a few questions regarding this: 
- Would this be functionality that adafruit would be interested in merging into this repo (or separating into a completely separate driver)?
- Would people who are more familiar with either the specific module or radio use in general be able to provide feedback on my code as I write it, since I am still relatively new to radio hardware and CircuitPython? 

I plan to update this issue as I go, as well as create a PR when the code is in a more mature state.

brentru, 2020-12-16:
> Would this be functionality that adafruit would be interested in merging into this repo (or separating into a completely separate driver)?

Sure - I'd like to see how you're implementing it in user-code to determine if we should organize the files further.

Could you add an example to that fork?

> Would people who are more familiar with either the specific module or radio use in general be able to provide feedback on my code as I write it, since I am still relatively new to radio hardware and CircuitPython?

Yep! Please tag me when you open up a Pull Request.

jerryneedell, 2020-12-16:
Just curious -- If this works would you expect it to be possible for an RFM9x to talk to an RFM69 if both configured for FSK?

Michael-Equi, 2021-11-12:
@lgnashold Did you ever get your FSK/OOK fork working? I need FSK for a satellite project and I am not getting any errors after running your code on a PyCubed board with an RFM98 but it does not seem to be transmitting anything. 

I tried with both `self.modulation_type = _OOK_MODULATION` and `self.modulation_type = _FSK_MODULATION`

jerryneedell, 2024-03-08:
FYI - I do have FSK working on teh RFM9x and it can communicate with an RFM69 module.  I am working on a new library that combines the RFM69 and RFM9x libraries and adds some options.( FSK for RFM9x,  headerless (no radiohead header).
It is still a "Work In Progress"  https://github.com/jerryneedell/CircuitPython_RFM

jerryneedell, 2025-02-17:
Implemented in the Adafruit_CircuitPython_RFM library
https://github.com/adafruit/Adafruit_CircuitPython_RFM.git
clapping here since it will add too much code to this library for the smaller M0 boards (feather_m0_rfm9x)

closed 2025-02-17 by jerryneedell