adafruit/Adafruit_CircuitPython_RFM9x #61: Check wiring error when the wiring is good

joeyb197, 2021-07-07:
Hi, 

I am trying to make two raspberry pi communicate through LoRa (SX127x). (https://media.digikey.com/pdf/Data%20Sheets/Semtech%20PDFs/SX1272MB2DAS_HM.PDF)

When I try the following code, I get the following error : 

Code : 
**import digitalio
import board
import busio
import adafruit_rfm9x

RADIO_FREQ_MHZ = 915.0
CS = digitalio.DigitalInOut(board.CE1)
RESET = digitalio.DigitalInOut(board.D25)
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, RADIO_FREQ_MHZ)**

Error : 
**Traceback (most recent call last):
File "/home/pi/Downloads/test.py", line 10, in 
rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, RADIO_FREQ_MHZ)
File "/usr/local/lib/python3.7/dist-packages/adafruit_rfm9x.py", line 250, in __init__
"Failed to find rfm9x with expected version -- check wiring"
RuntimeError: Failed to find rfm9x with expected version -- check wiring**

The wiring is good (double-triple-quadruple checked)

I tried switching CE1 and CE2 and GPIO's, but it doesn't seem to work.

Now I know that this is a RFM library, but the RFM is build on SX127x's chips isn't it?

Any help would be greatly appreciated!

jerryneedell, 2021-07-07:
@joeyb197 please post this to the support forum . https://forums.adafruit.com/viewforum.php?f=60

This is most likely a configuration or wiring issue and not a library bug.

joeyb197, 2021-07-07:
Thanks! I'll do it!

closed 2021-07-08 by tannewt