adafruit/Adafruit_CircuitPython_RFM9x #67: Enhancement: Switch time.monotonic() to supervisor.ticks_ms()

mrdalgaard, 2021-11-30:
As many RFM9x projects are likely to be in a remote location, and running for long'ish periods, the accuracy of time.monotonic() will over time degrade into the seconds range, making send_with_ack, and receive timeouts longer then desired.

self.ack_wait = 0.5 would effectively become self.ack_wait = 1
self.receive_timeout = 0.5 would become self.receive_timeout = 1
This would become worse over time. (higher values)

Switching to supervisor.ticks_ms() would solve this, however I'm unsure if its available on all platforms yet.

jerryneedell, 2021-11-30:
supervisor appears to be supported by most builds -- I'll do some checking, but this may be a good way to go
https://circuitpython.readthedocs.io/en/latest/shared-bindings/supervisor/index.html#supervisor.ticks_ms

As discussed on discord, I was hoping "deep sleep" would also help, but it is not clear if it will.

mrdalgaard, 2021-11-30:
> supervisor appears to be supported by most builds -- I'll do some checking, but this may be a good way to go [...]

I have working code, with time.monotonic() replaced with supervisor.ticks_ms(), and the diff function mentioned in your link.
Let me know if you would like me to send it.

Unfortunately the whole Pull Request, pylint, and black is still a bit to complicated for me.

jerryneedell, 2021-11-30:
@mrdalgaard Feel free to post your code here -- I was just looking at what it would involve.

mrdalgaard, 2021-11-30:
Heres the diff, just to keep it brief:

```
15a16
> import supervisor
102a104,107
> # supervisor.ticks_ms() contants
> _TICKS_PERIOD = const(1<<29)
> _TICKS_MAX = const(_TICKS_PERIOD-1)
> _TICKS_HALFPERIOD = const(_TICKS_PERIOD//2)
343a349,354
>     def ticks_diff(self, ticks1, ticks2):
>         "Compute the signed difference between two ticks values, assuming that they are within 2**28 ticks"
>         diff = (ticks1 - ticks2) & _TICKS_MAX
>         diff = ((diff + _TICKS_HALFPERIOD) & _TICKS_MAX) - _TICKS_HALFPERIOD
>         return diff
>
704c715
<         start = time.monotonic()
---
>         start = supervisor.ticks_ms()
707c718
<             if (time.monotonic() - start) >= self.xmit_timeout:
---
>             if self.ticks_diff(supervisor.ticks_ms(), start) >= self.xmit_timeout * 1000:
784c795
<             start = time.monotonic()
---
>             start = supervisor.ticks_ms()
787c798
<                 if (time.monotonic() - start) >= timeout:
---
>                 if self.ticks_diff(supervisor.ticks_ms(), start) >= timeout * 1000:

```

jerryneedell, 2021-11-30:
@mrdalgaard Thanks -- looks good. I was at first concerned about mixing the float values and the int values but python seems to be pretty good at that. I'll do some testing and create a PR for discussion in the next few days. Thanks for pronging this up.

jerryneedell, 2021-12-01:
hmm -- ran into a snag ... The implementation of "Blinka" that supports using Circuitpython libraries on Raspberry Pi's does not support "supervisor" .... 
I'll could try adding a try/except on the import of supervisor and if it fails, fall back to using time.monnotonic()... I'll think about this a bit.
Any other suggestions welcome.

jerryneedell, 2021-12-01:
I have a kludge working on the Raspberry py and regular MCU's -- I'm still looking for a better way. Here is my working version. I'm not ready to commit it yet.
```
# SPDX-FileCopyrightText: 2017 Tony DiCola for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""
`adafruit_rfm9x`
====================================================

CircuitPython module for the RFM95/6/7/8 LoRa 433/915mhz radio modules.  This is
adapted from the Radiohead library RF95 code from:
http: www.airspayce.com/mikem/arduino/RadioHead/

* Author(s): Tony DiCola, Jerry Needell
"""
import random
import time
import adafruit_bus_device.spi_device as spidev
from micropython import const
import supervisor

__version__ = "0.0.0-auto.0"
__repo__ = "https://github.com/adafruit/Adafruit_CircuitPython_RFM9x.git"

# Internal constants:
# Register names (FSK Mode even though we use LoRa instead, from table 85)
_RH_RF95_REG_00_FIFO = const(0x00)
_RH_RF95_REG_01_OP_MODE = const(0x01)
_RH_RF95_REG_06_FRF_MSB = const(0x06)
_RH_RF95_REG_07_FRF_MID = const(0x07)
_RH_RF95_REG_08_FRF_LSB = const(0x08)
_RH_RF95_REG_09_PA_CONFIG = const(0x09)
_RH_RF95_REG_0A_PA_RAMP = const(0x0A)
_RH_RF95_REG_0B_OCP = const(0x0B)
_RH_RF95_REG_0C_LNA = const(0x0C)
_RH_RF95_REG_0D_FIFO_ADDR_PTR = const(0x0D)
_RH_RF95_REG_0E_FIFO_TX_BASE_ADDR = const(0x0E)
_RH_RF95_REG_0F_FIFO_RX_BASE_ADDR = const(0x0F)
_RH_RF95_REG_10_FIFO_RX_CURRENT_ADDR = const(0x10)
_RH_RF95_REG_11_IRQ_FLAGS_MASK = const(0x11)
_RH_RF95_REG_12_IRQ_FLAGS = const(0x12)
_RH_RF95_REG_13_RX_NB_BYTES = const(0x13)
_RH_RF95_REG_14_RX_HEADER_CNT_VALUE_MSB = const(0x14)
_RH_RF95_REG_15_RX_HEADER_CNT_VALUE_LSB = const(0x15)
_RH_RF95_REG_16_RX_PACKET_CNT_VALUE_MSB = const(0x16)
_RH_RF95_REG_17_RX_PACKET_CNT_VALUE_LSB = const(0x17)
_RH_RF95_REG_18_MODEM_STAT = const(0x18)
_RH_RF95_REG_19_PKT_SNR_VALUE = const(0x19)
_RH_RF95_REG_1A_PKT_RSSI_VALUE = const(0x1A)
_RH_RF95_REG_1B_RSSI_VALUE = const(0x1B)
_RH_RF95_REG_1C_HOP_CHANNEL = const(0x1C)
_RH_RF95_REG_1D_MODEM_CONFIG1 = const(0x1D)
_RH_RF95_REG_1E_MODEM_CONFIG2 = const(0x1E)
_RH_RF95_REG_1F_SYMB_TIMEOUT_LSB = const(0x1F)
_RH_RF95_REG_20_PREAMBLE_MSB = const(0x20)
_RH_RF95_REG_21_PREAMBLE_LSB = const(0x21)
_RH_RF95_REG_22_PAYLOAD_LENGTH = const(0x22)
_RH_RF95_REG_23_MAX_PAYLOAD_LENGTH = const(0x23)
_RH_RF95_REG_24_HOP_PERIOD = const(0x24)
_RH_RF95_REG_25_FIFO_RX_BYTE_ADDR = const(0x25)
_RH_RF95_REG_26_MODEM_CONFIG3 = const(0x26)

_RH_RF95_REG_40_DIO_MAPPING1 = const(0x40)
_RH_RF95_REG_41_DIO_MAPPING2 = const(0x41)
_RH_RF95_REG_42_VERSION = const(0x42)

_RH_RF95_REG_4B_TCXO = const(0x4B)
_RH_RF95_REG_4D_PA_DAC = const(0x4D)
_RH_RF95_REG_5B_FORMER_TEMP = const(0x5B)
[... 779 lines cut ...]
                            node=packet[0],
                            identifier=packet[2],
                            flags=(packet[3] | _RH_FLAGS_ACK),
                        )
                        # reject Retries if we have seen this idetifier from this source before
                        if (self.seen_ids[packet[1]] == packet[2]) and (
                            packet[3] & _RH_FLAGS_RETRY
                        ):
                            packet = None
                        else:  # save the packet identifier for this source
                            self.seen_ids[packet[1]] = packet[2]
                    if (
                        not with_header and packet is not None
                    ):  # skip the header if not wanted
                        packet = packet[4:]
        # Listen again if necessary and return the result packet.
        if keep_listening:
            self.listen()
        else:
            # Enter idle mode to stop receiving other packets.
            self.idle()
        # Clear interrupt.
        self._write_u8(_RH_RF95_REG_12_IRQ_FLAGS, 0xFF)
        return packet
```

jerryneedell, 2021-12-01:
I have a kludge working on the Pi and regular MCUs... I'm not ready to commit it yet, but here it is if you want to have a look and comment. 
```
# SPDX-FileCopyrightText: 2017 Tony DiCola for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""
`adafruit_rfm9x`
====================================================

CircuitPython module for the RFM95/6/7/8 LoRa 433/915mhz radio modules.  This is
adapted from the Radiohead library RF95 code from:
http: www.airspayce.com/mikem/arduino/RadioHead/

* Author(s): Tony DiCola, Jerry Needell
"""
import random
import time
import adafruit_bus_device.spi_device as spidev
from micropython import const
try:
    import supervisor
    no_supervisor = False
except:
    import time
    no_supervisor = True

__version__ = "0.0.0-auto.0"
__repo__ = "https://github.com/adafruit/Adafruit_CircuitPython_RFM9x.git"

# Internal constants:
# Register names (FSK Mode even though we use LoRa instead, from table 85)
_RH_RF95_REG_00_FIFO = const(0x00)
_RH_RF95_REG_01_OP_MODE = const(0x01)
_RH_RF95_REG_06_FRF_MSB = const(0x06)
_RH_RF95_REG_07_FRF_MID = const(0x07)
_RH_RF95_REG_08_FRF_LSB = const(0x08)
_RH_RF95_REG_09_PA_CONFIG = const(0x09)
_RH_RF95_REG_0A_PA_RAMP = const(0x0A)
_RH_RF95_REG_0B_OCP = const(0x0B)
_RH_RF95_REG_0C_LNA = const(0x0C)
_RH_RF95_REG_0D_FIFO_ADDR_PTR = const(0x0D)
_RH_RF95_REG_0E_FIFO_TX_BASE_ADDR = const(0x0E)
_RH_RF95_REG_0F_FIFO_RX_BASE_ADDR = const(0x0F)
_RH_RF95_REG_10_FIFO_RX_CURRENT_ADDR = const(0x10)
_RH_RF95_REG_11_IRQ_FLAGS_MASK = const(0x11)
_RH_RF95_REG_12_IRQ_FLAGS = const(0x12)
_RH_RF95_REG_13_RX_NB_BYTES = const(0x13)
_RH_RF95_REG_14_RX_HEADER_CNT_VALUE_MSB = const(0x14)
_RH_RF95_REG_15_RX_HEADER_CNT_VALUE_LSB = const(0x15)
_RH_RF95_REG_16_RX_PACKET_CNT_VALUE_MSB = const(0x16)
_RH_RF95_REG_17_RX_PACKET_CNT_VALUE_LSB = const(0x17)
_RH_RF95_REG_18_MODEM_STAT = const(0x18)
_RH_RF95_REG_19_PKT_SNR_VALUE = const(0x19)
_RH_RF95_REG_1A_PKT_RSSI_VALUE = const(0x1A)
_RH_RF95_REG_1B_RSSI_VALUE = const(0x1B)
_RH_RF95_REG_1C_HOP_CHANNEL = const(0x1C)
_RH_RF95_REG_1D_MODEM_CONFIG1 = const(0x1D)
_RH_RF95_REG_1E_MODEM_CONFIG2 = const(0x1E)
_RH_RF95_REG_1F_SYMB_TIMEOUT_LSB = const(0x1F)
_RH_RF95_REG_20_PREAMBLE_MSB = const(0x20)
_RH_RF95_REG_21_PREAMBLE_LSB = const(0x21)
_RH_RF95_REG_22_PAYLOAD_LENGTH = const(0x22)
_RH_RF95_REG_23_MAX_PAYLOAD_LENGTH = const(0x23)
_RH_RF95_REG_24_HOP_PERIOD = const(0x24)
_RH_RF95_REG_25_FIFO_RX_BYTE_ADDR = const(0x25)
_RH_RF95_REG_26_MODEM_CONFIG3 = const(0x26)

_RH_RF95_REG_40_DIO_MAPPING1 = const(0x40)
_RH_RF95_REG_41_DIO_MAPPING2 = const(0x41)
[... 798 lines cut ...]
                            node=packet[0],
                            identifier=packet[2],
                            flags=(packet[3] | _RH_FLAGS_ACK),
                        )
                        # reject Retries if we have seen this idetifier from this source before
                        if (self.seen_ids[packet[1]] == packet[2]) and (
                            packet[3] & _RH_FLAGS_RETRY
                        ):
                            packet = None
                        else:  # save the packet identifier for this source
                            self.seen_ids[packet[1]] = packet[2]
                    if (
                        not with_header and packet is not None
                    ):  # skip the header if not wanted
                        packet = packet[4:]
        # Listen again if necessary and return the result packet.
        if keep_listening:
            self.listen()
        else:
            # Enter idle mode to stop receiving other packets.
            self.idle()
        # Clear interrupt.
        self._write_u8(_RH_RF95_REG_12_IRQ_FLAGS, 0xFF)
        return packet
```

mrdalgaard, 2021-12-01:
Since we're already checking for supervisor.ticks_ms, would checking for time.monotonic_ns as the first fallback also make sense? Or is that just overcomplicating things..?

The issue faced is repeated across many modules using time.monotonic() for timeouts, just to mention a few:
https://github.com/adafruit/Adafruit_CircuitPython_SI4713/blob/main/adafruit_si4713.py
https://github.com/adafruit/Adafruit_CircuitPython_GPS/blob/main/adafruit_gps.py
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/main/adafruit_dht.py
https://github.com/adafruit/Adafruit_CircuitPython_PN532/blob/main/adafruit_pn532/i2c.py

Even though the RFM library will likely experience the problem the most (the timeout here is used for non wired signals, and with a fairly short timeout), maybe a generalized approach would make sense. Either way, I'm happy if it is solved in the RFM9x lib.

jerryneedell, 2021-12-01:
the time.monotonic_ns is not supported on boards with longints so this will be a problem for some MCUs.
https://circuitpython.readthedocs.io/en/latest/shared-bindings/time/#time.monotonic_ns <https://circuitpython.readthedocs.io/en/latest/shared-bindings/time/#time.monotonic_ns>

I agree that this may need to be considered for other libraries as well.

jerryneedell, 2021-12-01:
Also, time.monotonic() does not have the same accuracy loss on the Raspberry Pi so it should not be an issue there.

jerryneedell, 2021-12-01:
@mrdalgaard PR #68 created -- please review and comment.

PR #68 (2021-12-03): use supervisor.ticks_ms when available instead of time.monotonic()
Thank you to https://github.com/mrdalgaard for suggesting this.
Closes: #67
This change will use supervisor.ticks_ms() instead of time.monotonic() for the timeout checks.
This will avoid potential issues with time.monotonic() losing accuracy with time.

For the Raspberry Pi implementation via Blinka, the supervisor module is not preset so time.monotonic() is still used but the Pi does not suffer for the same degradation of the accuracy of time.monotonic() so it should not be an issue there.

I have tested this on a feather_rp20 [... cut]
closed 2021-12-03 by jerryneedell