adafruit/Adafruit_CircuitPython_RFM9x #71: rfm9x.send causes ble disconnect

dewalex, 2022-03-17:
I have been using the bluefruit M0 for a project with LoRa but I really needed more ram, so I am trying to switch to the nRF52840.  The M0 was successfully sending and receiving data to and from a custom iOS app over SPI and I'm using uart on the nRF52.  

Using this library is causing the ble connection to terminate.  See the below for an example, running as is and then commenting out the ```rfm9x.send()``` line.  Is this a known issue?

```

import time
import busio
import board
from digitalio import DigitalInOut, Direction, Pull
from analogio import AnalogIn

from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

import adafruit_rfm9x

# Define radio parameters.
RADIO_FREQ_MHZ = 915.0  # Frequency of radio in Mhz.

# Define pins
CS = DigitalInOut(board.D10)  # "B" on the board
RESET = DigitalInOut(board.D11) # "A" on board

# Define the onboard LED
LED = DigitalInOut(board.D3)
LED.direction = Direction.OUTPUT
LED.value = True

# Initialize SPI bus.
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)

# Initialze RFM radio
rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, RADIO_FREQ_MHZ)
rfm9x.tx_power = 20

rfm9x.signal_bandwidth = 250000
rfm9x.coding_rate = 5
rfm9x.spreading_factor = 11
rfm9x.enable_crc = False

ble = BLERadio()
uart = UARTService()
advertisement = ProvideServicesAdvertisement(uart)

while True:
    ble.start_advertising(advertisement)
    print("Waiting to connect")
    while not ble.connected:
        pass
    print("Connected")
    while ble.connected:
        s = uart.read()
        if s:
            try:
                rfm9x.send(bytes(str(s)[-2], "utf-8"))
                result = str(eval(s))
            except Exception as e:
                result = repr(e)

            uart.write(result.encode("utf-8"))
```

I also am having really strange issues where the uart is sending everything that I ```print()``` in my code over to the iphone and not just what i send with uart.write().  That may or may not be a related issue to the library.

0xabe1, 2023-04-12:
Well, I know it's been a year since you posted, but I'm having the same issue. Any chance you found a solution?

It looks like this guy is too (assuming you're different people), so there's likely some problem with the two board combo:

https://forums.adafruit.com/viewtopic.php?t=174471

I'm using Zephyr and coding in C, so it's not specific to circuitpython (though they likely share a BLE stack). 

My boards have Ultimate GPS wings on them too. Do you have additional modules?

My best guess right now is that it's either a power issue or an RF interference problem. What I noticed is that if I bring the TX power down below where the RFM95's PA gets enabled that my BLE connection suddenly gets very stable. I think that's 17 dB or 15 dB. Looks like you're at 20 dB. 

I also noticed different combinations of connections changes how high I can set the RFM95 power. USB only - 19 dB, USB+LiPo - 17 dB, USB+LiPo+JTAG - 15 dB. 

I haven't had time to dig any deeper than the TX power. Sadly that's killing my range. Getting the full 6+dB more would do great things.

dewalex, 2023-04-13:
Unfortunately I didn't find a solution for this.  I just ended up switching back to the M0 and writing an exception that resets on memory error :[

I don't have any extra modules on my setup.  I have the Bluefruit + LoRa as a receiver for a LoRa+GPS as a tracker on my dog.  Can't afford any less power as I'm only getting a max of 400m in the woods using a SF of 11 and BW of 12500, which was as high SF and low BW I could go without interfering with GPS passing data to the board.

0xabe1, 2023-04-18:
I'm fairly certain I found the problem. I have a dozen functional assemblies now. Posted to the forum since I think it's hardware/software combo related, but I haven't seen a response from Adafruit yet. 

https://forums.adafruit.com/viewtopic.php?t=200678&sid=83c0a9b874edb4fea41ea266bb6fd895

jerryneedell, 2025-02-17:
Closing since this does not appear to be an issue with this library -- Please re-open if that is not correct.

closed 2025-02-17 by jerryneedell