adafruit/Adafruit_CircuitPython_RFM9x #75: SPI Bus interference from SDCard

FoamyGuy, 2022-05-09:
In testing a recent PR I noticed that using the RFM9x Radio FeatherWing on a PyGamer (7.3.0-beta.2) when I have an SDCard plugged in the radio doesn't behave correctly. It will cause the same message to get received over and over again rapidly, as well as messages after the first one not being received by the other side. 

Removing the SDCard and then re-running the same examples on the same devices then works successfully and behaves as expected when the SDCard is not plugged in. 

My best guess is that something internally within the core is perhaps initializing or starting communications with the SDCard on the SPI bus and that is causing interference when the RFM9x radio tries to communicate on the same SPI bus.

bablokb, 2024-02-16:
Now this is quite old, but I also have problems with a SDCard and a second device (display) on the SPI bus. One possible source of the problem can be missing pullups on the CS lines: if you don't have pullups, then before all devices are initialized properly the CS are potentially floating and the other devices will "see" commands not meant for them.

jerryneedell, 2024-02-23:
@FoamyGuy Did you ever resolve this? Should it remain open?