adafruit/Adafruit_CircuitPython_RFM9x #88: RFM95x micropython and raspberry pico

JavaldeUB, 2023-05-03:
Hi,
I'm building up a loRA P2P net with a raspberry pico sending data to a RP4. At the moment I'm stuck with the pico and blinka. I'm able to wire the pico to the rfm9x and make it work with the circuitpython firmware in the pico. However, with the same wiring, I try use blinka with micropython and the "Failed to find rfm9x with expected version -- check wiring" error pops up.

The wiring:
3.3V OUTPUT----->3.3V
GND-------------->GND
GP18------------->SCK
GP19------------->MOSI
GP16------------->MISO
GP8-------------->CS
GP9-------------->RST

 I would like to have this running so I can use the two cores of the Pico.
I do not know if you have arecordd on this problem or encountered something similar.
All the best,
Javi

JavaldeUB, 2023-05-03:
Findings.
I've left the wiring as it is and run the following code:

import time
import busio
import digitalio
import board
import microcontroller
import adafruit_rfm9x

# Board LED
led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

# Create library object using our bus SPI port for radio
spi = busio.SPI(board.GP18, MOSI=board.GP19, MISO=board.GP16)
# RFM9x Breakout Pinouts
cs = digitalio.DigitalInOut(board.GP8)
rst = digitalio.DigitalInOut(board.GP9)

cs.direction = digitalio.Direction.OUTPUT

buffer = bytearray(1)

spi.try_lock()
cs.value = False
spi.write(b'\x42')
spi.readinto(buffer)
cs.value = True
spi.unlock()

print("Version = ",buffer[0])

And I've got a proper answer: Version 18. Which is the value the driver is waiting for.
However, when it runs the __init_ in the RFM9x.py I've got Version 0. My thinking is that the spi does the job but at some point the driver does not lock the SPI or does not toggle with the CS pin.
Any idea?
Javi

jerryneedell, 2024-01-04:
Is this an issue with Blinka or with this library?  I have not run into this with Blinka running on Raspberry Pis with RaspiOS. I have not tried it with MicroPython.

closed 2024-02-23 by jerryneedell

Patcybermind, 2025-03-09:
hey @JavaldeUB did you find out how to fix it? im having the same problem

jerryneedell, 2025-03-14:
> hey [@JavaldeUB](https://github.com/JavaldeUB) did you find out how to fix it? im having the same problem

I think it would be best for you to open a new issue and clearly explain the problems you are having.