adafruit/Adafruit_CircuitPython_RFM9x #89: frequency_mhz return value is not the same as the set frequency value due to floating point division

BNNorman, 2023-08-13:
When setting the frequency then reading back to check it was set the value returned for, say, 868.1 can be 868.099.... due to floating point calculations. The actual register values read back are, obviously,identical to the values set 

On line 481 I found that returning round(frequency,2) fixed the problem for me with LoraWAN frequencies such as 868.1, 868.3, 868.5 etc

jerryneedell, 2024-01-07:
@BNNorman  Sorry for the slow response.
I don't think it is correct to consider this a "bug".  The frequency can only be set as specified by Page 109 of:
https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf
and it can only be set in steps 61.03515625 Hz  (32,000,000/2^19) so values like 868.1 MHz cannot be precisely achieved.
The value you see is the actual frequency being set. That is done in line 488 https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/main/adafruit_rfm9x.py#L488

I am going to close this issue. Please reopen it if you don't agree with my assessment.

jerryneedell, 2024-01-07:
oops - meant to close with the above comment

closed 2024-01-07 by jerryneedell