adafruit/Adafruit_CircuitPython_RFM9x #91: RFM95W Bonnet on RPi5

martin-grundlingh, 2024-01-28:
I am new to many parts of single board computing, Linux and add ons which is why I opted to go for a PHAT instead of individual parts to get LoRA comms up and running. I am running into some issues which are challenging my knowledge and tshoot abilities.

I have 2 x Adafruit RFM95W Bonnet units:
1. Installed on a RPi Zero 2 W and working as expected
2. Installed on a RPi5 8GB and is not working as expected

I followed the instructions from the Wiki on both. It worked on the Pi0 but not the Pi5. Running the scupplied script results in the following error:

```

Traceback (most recent call last):
  File "/home/admin/Desktop/Adafruit/rfm9x_check.py", line 48, in <module>
    CS = DigitalInOut(board.CE1)
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/digitalio.py", line 185, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/digitalio.py", line 215, in direction
    self._pin.init(mode=Pin.IN)
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod/libgpiod_pin_1_x.py", line 92, in init
    self._line.request(
OSError: [Errno 16] Device or resource busy

```

I am only starting with Python so troubleshooting has been mostly Google searches, but I fear I might need a better answer from someone in the know. I tried reading through all of the libraries in question. It looks like the libraries referenced do not accomodate for the Pi5, or does so incorrectly (not sure if it has anything to do with the GPIO having their own dedicated controller now?).

Can anyone confirm if the Bonnet has been tested successfully on a RPi5? If not, how does one typically engage with ADAFRUIT support staff responsible for scripts packaged with their products?

makermelissa, 2024-01-29:
Please check out the learn guide for how to fix this: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices#reassigning-or-disabling-the-spi-chip-enable-lines-3097985

martin-grundlingh, 2024-01-29:
Hi - can you please be a little more specific. As mentioned, I’m not well versed in the subjects. 

Also, what would I need to uninstall / reinstall?

jerryneedell, 2024-01-30:
@makermelissa So, If I am understanding this correctly, for something like the RFM95 Bonnet, which has the CS hardwired to CE1, I would use
```
sudo -E env PATH=$PATH python3 raspi-spi-reassign.py --ce0=disabled --ce1=disabled
```
to disable OS use of the CE0 and CE1 pins. Is that correct?

Also, - this only appears to be an issue for the Pi 5 -- I have been using these bonnets for years on other Pis....

jerryneedell, 2024-02-23:
On a Pi5 I ran into the CE1 inuse error with the rfm9x_rpi_simpletest.py example.
I downloaded the raspi-spi-reassign.py script from the guide noted above  and ran
```sudo -E env PATH=$PATH python3 raspi-spi-reassign.py --ce0=disabled --ce1=disabled``` rebooted and now the rfm9x_rpi_simpletest example runs as expected.

I will close this issue. Please re-open it if you still have problems.

closed 2024-02-23 by jerryneedell