adafruit/Adafruit_CircuitPython_RFM9x #94: spreading_factor = 6 does not work on Ra-02 (SX1278)

stanelie, 2024-12-02:
I can transmit and receive data if I set the spreading factor anywhere between 7 (default) and 12 on both transmitter and receiver, but setting it to 6 fails to transmit/receive anything.

I'm using Adafruit CircuitPython 9.2.0 on 2024-10-28 on a Waveshare ESP32-S3-Zero with ESP32S3, with a Ra-02 LoRa module (https://www.aliexpress.com/item/32965792976.html)

jerryneedell, 2024-12-02:
Ugh! I think this relates to an old issue https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/issues/70 that has unfortunately not been resolved.

I'll look into it in the CircuitPython_RFM library https://github.com/adafruit/Adafruit_CircuitPython_RFM. and see if I can reproduce and hopefully understand this. If a fix is found, it can also be "back ported" to this library.

stanelie, 2024-12-03:
I notice in     @spreading_factor.setter (line 632 of adafruit_rfm9x.py) that there is a difference when sf is set to 6. I've tried to set it so detection_optimize stays to 0x3 and detection threshold stays set to 0x0A, but it still doesn't work.
```
    def spreading_factor(self, val: Literal[6, 7, 8, 9, 10, 11, 12]) -> None:
        # Set spreading factor (set to 7 to match RadioHead Sf128).
        val = min(max(val, 6), 12)

        if val == 6:
            self.detection_optimize = 0x3
        else:
            self.detection_optimize = 0x3

        self._write_u8(_RH_RF95_DETECTION_THRESHOLD, 0x0A if val == 6 else 0x0A)
        self._write_u8(
            _RH_RF95_REG_1E_MODEM_CONFIG2,
            (
                (self._read_u8(_RH_RF95_REG_1E_MODEM_CONFIG2) & 0x0F)
                | ((val << 4) & 0xF0)
            ),
        )
```

jerryneedell, 2025-01-20:
As noted in the Adafruit_CIrcuitPython_RFM library issue https://github.com/adafruit/Adafruit_CircuitPython_RFM/issues/8,  According to the SX126 data sheet "Note With SF = 6 selected, implicit header mode is the only mode of operation possible."

This library assumes the header is in "explicit" mode. Several changes would be needed to implement "implicit" mode.

So I think that SF 7-12 are actually working OK, but for the higher SF especially 12, more experimentation on the timing parameters is needed due to the increase in the packet transmission and receive times. I'll be looking into that more.

jerryneedell, 2025-01-26:
Some progress on SF= 6...see https://github.com/adafruit/Adafruit_CircuitPython_RFM/issues/8#issuecomment-2614554616

jerryneedell, 2025-01-28:
see https://github.com/adafruit/Adafruit_CircuitPython_RFM/pull/10
SF=6 is working in the RFM library (except for "reliable datagram mode")

stanelie, 2025-01-28:
Thanks!
I'll test it next time I work with the Ra-02.

jerryneedell, 2025-01-28:
> Thanks! I'll test it next time I work with the Ra-02.

Just to be clear, the PR is to the Adafruit_CircuitPython_RFM library.
I'm not sure if the changes will be made to the Adafruit_CiicruitPython_RFM9x library since it add code to the already overtaxed M0 boards...

jerryneedell, 2025-01-28:
> Oh, right, so, it doesn't change anything for me and my Ra-02. Oh well, I meant to test the rfm anyway, and I got a few chips for that. It's on my list.
> 
> Thanks again!

If the Adafruit_CircuitPython_RFM9x library works on you Ra-O2, then the RFM library should work as well.  Please let me know if it does not!