adafruit/Adafruit_CircuitPython_RFM9x #96: missing documentation on CircutPython website

codebreaker133, 2025-03-27:
On the CircutPython website https://docs.circuitpython.org/projects/rfm9x/en/latest/api.html#adafruit_rfm9x.RFM9x.last_rssi there are a few missing functions and there definition's; low_datarate_optimize() (not listed or defined the use case), 
radiohead(), 
enable_address_filter(),
 fsk_node_address() (what is the difference from node() ?)  
fsk_broadcast_address(), 
rfm.encryption_key(), 
rfm9xfsk.RFM9xFSK() (FSK is not defined or explained in any capacity), 
rfm.payload_ready(), 
spreading_factor() (it dose not say when or how you can use sf 6, I know you can only use it in implicit header mode but that's only because I read the data sheet), 
I believe there are more commands not listed in the docs or there sintax have changed, it would be beneficial to go through the examples here in the GitHub and compare them to what's on the circutpython docs to make sure that there accurate, I am new to this module and have relatively little experience although I do have a project that I am currently working on, but it is hard to understand the changes that have been made post December of 24 because there is limited docs (that I can find) for witch commands were changed and how.

I think all the examples shown no longer work due to changes in the way you reference the module when first setting up the radio as an example;

https://github.com/adafruit/Adafruit_CircuitPython_RFM/blob/8f0b80e279201b24d204a57042e97c70b6229f56/examples/rfm_simpletest.py#L35 this line is different between the GitHub docs and CircutPython docs

jerryneedell, 2025-03-27:
These are new features in the The Adafruit_CircuitPython_RFM library [-](https://github.com/adafruit/Adafruit_CircuitPython_RFM) - The Adafruit_CIrcuitPythonRFM9x library is unchanged and will remain available.

See https://docs.circuitpython.org/projects/rfm/en/latest/api.html  for the RFM library Docs.
I am working on  new guide to better explain the  the RFM library. It is a "work in progress".
My apologies for the delay.

There are new examples for the RFM library in https://github.com/adafruit/Adafruit_CircuitPython_RFM/tree/main/examples

codebreaker133, 2025-03-27:
I really appreciate the swift reply and the hard work you are putting into this library this is my first major project in the microcontroller world and im glad to see such an active and helpful community, thanks for pointing me in the right direction and I can't wait for the new RFM library doc's, Sorry about the bluntness of my reply just wanted to make sure you knew what had or hadn't changed asap.

jerryneedell, 2025-03-27:
Please keep using and testing the library. Feel free to open issues in https://github.com/adafruit/Adafruit_CircuitPython_RFM/issues  if you have any problems.  
It is taking a long time to make good examples because I keep going down "rabbit holes" when I think I have a good example....